Ac6Embedded/vscode-zephyr-workbench #1: Unable to create application from standalone example

beriberikix, 2024-07-23:
https://github.com/golioth/reference-design-template

Hello! I'm testing out the extension with our Reference Design template. I successfully created a workspace by pointing to the repo @ main. When I went to create an application, it couldn't find any of the board files. It also was looking for Samples, which this repo is actually the entire application.

I'm guessing the main issue is that the extension doesn't handle this style of [topology](https://docs.zephyrproject.org/latest/develop/west/workspaces.html#topologies-supported)?

KevinTangDev, 2024-07-24:
Hi,
Indeed, the extension doesn't support this topology yet. I'll work on it asap! 
By the way, thank you for testing the extension and your feedback.

KevinTangDev, 2024-07-25:
A beta version is published and should solve the issue.
The west configuration manifest is now parsed to get the topology. I could find the boards and samples from your repo.
You might need to create the workspace again to be able to have the changes.

closed 2024-08-17 by RoyAc6