Ac6Embedded/vscode-zephyr-workbench #3: Fails to create west workspace from Repository

tiborkiss, 2024-10-15:
In a Windows 11 Pro machine I can initialize a workspace with minimal template and then everything works, including the debugging.
But when I tried to initialize a workspace from Repository (the second source location) fails with 

```
Updating files: 100% (35876/35876), done.
Traceback (most recent call last):
  File "C:\Users\Hades\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 853, in move   
    os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'c:\\w\\ze-ws\\zephyr\\.west\\manifest-tmp' -> 'c:\\w\\ze-ws\\zephyr\\zephyr'
...
  File "C:\Users\Hades\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 634, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\Hades\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 632, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'c:\\w\\ze-ws\\zephyr\\.west\\manifest-tmp\\.git\\objects\\pack\\pack-26b6145f5467869ec3837a4b90952956c96cd946.idx'
....
  File "C:\Users\Hades\.zinstaller\.venv\Lib\site-packages\west\configuration.py", line 227, in _get    
    for cf in self._whence(configfile):
              ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Hades\.zinstaller\.venv\Lib\site-packages\west\configuration.py", line 248, in _whence 
    raise MalformedConfig('local configuration file not found')
west.configuration.MalformedConfig: local configuration file not found

 *  The terminal process "cmd.exe /d /c call C:\Users\Hades\.zinstaller\env.bat && west update" terminated with exit code: 1. 
```
This Access is denied message is confusing and in the filesystem I don't see any file or folder with other owners and the current user has admin rights.
I think the last message is more relevant, some local configuration file is not found.

tiborkiss, 2024-10-16:
It happens in another PC too.
If I navigate to .zinstaller and activate venv to have west, `west update` will raise the same error.
If i completely remove the zephyr-v3.7.0 folder and I try `west init -m https://github.com/zephyrproject-rtos/zephyr --mr v3.7.0 zephyr-v3.7.0`, I still get error, somewhat similar error.

```
Updating files: 100% (35876/35876), done.
Traceback (most recent call last):
  File "C:\Users\Tibor\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 853, in move
    os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'C:\\w\\EASY6\\ze-ws\\zephyr-v3.7.0\\.west\\manifest-tmp' -> 'C:\\w\\EASY6\\ze-ws\\zephyr-v3.7.0\\zephyr'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Tibor\.zinstaller\.venv\Scripts\west.exe\__main__.py", line 7, in <module>
  File "C:\Users\Tibor\.zinstaller\.venv\Lib\site-packages\west\app\main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "C:\Users\Tibor\.zinstaller\.venv\Lib\site-packages\west\app\main.py", line 244, in run
    self.run_command(argv, early_args)
  File "C:\Users\Tibor\.zinstaller\.venv\Lib\site-packages\west\app\main.py", line 503, in run_command
    self.run_builtin(args, unknown)
  File "C:\Users\Tibor\.zinstaller\.venv\Lib\site-packages\west\app\main.py", line 611, in run_builtin
    self.cmd.run(args, unknown, self.topdir,
  File "C:\Users\Tibor\.zinstaller\.venv\Lib\site-packages\west\commands.py", line 194, in run
    self.do_run(args, unknown)
  File "C:\Users\Tibor\.zinstaller\.venv\Lib\site-packages\west\app\project.py", line 224, in do_run
    topdir = self.bootstrap(args)
             ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tibor\.zinstaller\.venv\Lib\site-packages\west\app\project.py", line 339, in bootstrap
    shutil.move(os.fspath(tempdir), os.fspath(manifest_abspath))
  File "C:\Users\Tibor\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 871, in move
    rmtree(src)
  File "C:\Users\Tibor\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 787, in rmtree
    return _rmtree_unsafe(path, onerror)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tibor\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 629, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Tibor\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 629, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Tibor\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 629, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Tibor\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 634, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\Tibor\.zinstaller\tools\python\python-3.11.8.amd64\Lib\shutil.py", line 632, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'C:\\w\\EASY6\\ze-ws\\zephyr-v3.7.0\\.west\\manifest-tmp\\.git\\objects\\pack\\pack-02aa1e0ab4e85f314416ddd3221dd54be7d4016c.idx'
```

Using my own python venv with west, doesn't have this problem, therefore the env in .zinstaller could be the problem.

tiborkiss, 2024-10-16:
I also found the same type of issue here https://github.com/zephyrproject-rtos/west/issues/558

KevinTangDev, 2024-10-16:
Thank you for all the detailed tests you've done. I'll try reproduce the issue and let you know if I find a solution to avoid this workaround.

tiborkiss, 2024-10-18:
I am somewhat skeptical if this can be solved running within VSCode.
If I open a cmd terminal (outside of VSCode), but VSCode is still not closed. Then I try to initialize west repo in the folder which is also the workspace for VSCode, I get the same error. 
But if I close the VSCode, the same west initialization goes smoothly.
What does that means?! VSCode is scanning that folder where git under the control of west tool is doing it's file manipulation and the scanning by VSCode will get some locks over those files, then when those temporary files is going to be moved or deleted by west (or git), it will not be able to have access it.

If there is no explicit VSCode API to keep out locking anything from that folder, better don't try to solve and let the user execute the command outside while VSCode is closed. When I created a minimal repo, which is fast, this interference by surprise just did not occured, but what prevents us from happening in any of the scenarios where the end users wants to initialize a repo.
Simply, in Windows I would just hide the button which allows initialization from within VSCode and I would just leave the Import local button and give advice what to do outside VSCode.

KevinTangDev, 2024-10-23:
Yes, indeed, i didn't found any solution from our side to avoid the error. I don't know if there is any other extension that could lock the workspace folder. 
Anyway, we will write a warning note with the workaround in the documentation to address the issue.

closed 2024-10-25 by KevinTangDev