Ac6Embedded/vscode-zephyr-workbench #4: Q: How to make it recognize custom boards?

tiborkiss, 2024-10-17:
I created a custom board in a boards subdirectory relative to the application "hello_world". 
Board changing command couldn't find the new board name, but I found in the .vscode/settings.json the following two attributes:
```
"zephyr-workbench.board": 
"C_Cpp.default.compileCommands": 
```
which refers to a board name. Therefore manually I edited to be the new board name.

Then I applied "Clean" -> "Delete build" on the application view.
If I press "Open Terminal" on the same application, I have the .venv where I can build with the new board using west by adding `-- -DBOARD_ROOT=.` and it build correctly.
In the Applications view the new board name it appears with the correct name, but it cannot build by user interface because it cannot find the new board name.
The build folder at first look it is different. The build triggered from UI which fails, creates a folder with the name of the new board, while in other cases (before I created the new board) or how the west command is building now with the new board, does not create such folder.

Can you help me with some hints how to manage to recognize the new board, which otherwise I can build by in terminal by west command if I add -DBOARD_ROOT=. ?

RoyAc6, 2024-10-17:
We are working on creating an automated and a clean way for creating and manging custom boards.
For the time being the way I do it is in a seperate module, here is an example:
```
button
├── CMakeLists.txt
├── prj.conf
└── src
    └── main.c
mymodule
├── boards
│   └── ac6
│       └── myboard
│           ├── board.yml
└── zephyr
    └── module.yml
```
in module.yml:
```
build:
   settings:
     board_root: .
```
You must add the module to zephyr, either using west, by adding a new module (recommended when using west) or locally by adding in your application's cmake file:
```
set(EXTRA_ZEPHYR_MODULES fullpath/mymodule)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE}) 
```
NOTE: it should be before find_package
then you change the board in settings.json
```
    "zephyr-workbench.board": "myboard",
    "C_Cpp.default.compileCommands": "${workspaceFolder}/build/myboard/compile_commands.json",

```
Then build... it should work

PS: To answer your question, if you'd like to add custom arguments to the west command, you can set them in tasks.json -> West Build ->  "args"

tiborkiss, 2024-10-21:
It worked with `set(EXTRA_ZEPHYR_MODULES fullpath/mymodule)` in CMakeLists.txt. Unfortunately that fullpath in git repository is not looking good.
I wished would work by adding the module with west. Despite reading the documentation [Integrate modules in Zephyr build system](https://docs.zephyrproject.org/latest/develop/modules.html#integrate-modules-in-zephyr-build-system), I couldn't figure it out. Probably I don't have a west.yml in the apps or in the root when I create the project with Zephyr workbench. Editing the west.yml in  zephyr-v3.7.0/zephyr/west.yml I also don't like, because that will not be added to my git repo.

Here is how it looks like my VSCode workspace when using Zephyr workbench.

```
├── apps
│   └── hello_world
│       ├── CMakeLists.txt
│       ├── README.rst
│       ├── build
│       ├── prj.conf
│       ├── sample.yaml
│       └── src
├── sdk
│   └── zephyr-sdk-0.16.8
│       ├── ....
├── mycustom-boards
│   ├── boards
│   │   └── xyz-vendor
│   └── zephyr
│       └── module.yml
└── zephyr-v3.7.0
    ├── bootloader
    │   ├── ....
    ├── modules
    │   ├── ....
    ├── tools
    │   ├── ....
    └── zephyr
        ├── ....
        ├── submanifests
        ├── west.yml
        ├── ....
```
With west there are multiple layouts which can be used. When I just use west style of workspaces, there I have a west.yml in the top, which can be persisted in my git repo. Just that structure it did not worked with Zephyr workbench plugins, up until now.
I am trying to understand to which extend yes or no it is possible to follow west layouts within Zephyr workbench. Ideally, I wish to have something which adheres to some of the well known west workspace layouts. 
When I right click the application and click "Open Terminal", in that terminal I can work with west commands. Therefore to some extent, this setup supports working with west commands too.

Definitively I am looking forward with new developments.

RoyAc6, 2024-10-21:
Zephyr Workbench is indeed compatible with **west** and the various topologies that west supports. Adding modules via `EXTRA_ZEPHYR_MODULES` works fine for local folders (I have just retested it), but it is typically meant for situations where your module isn't managed by a Git repository or is intended to remain local. When your module is hosted on a Git repository (which is the recommended way), it should be managed using the west manifes.

You should set up your `west.yml` correctly. Here's an example of what your `west.yml` could look like to include your custom module, which would be persisted in your Git repository:

```
manifest:
  remotes:
    - name: zephyrproject
      url-base: https://github.com/zephyrproject-rtos
  projects:
    - name: zephyr
      repo-path: zephyr
      remote: zephyrproject
      revision: v3.7.0
      import:
        path-prefix: deps
        name-allowlist:
          - cmsis
          - cmsis-dsp
          - percepio
          - picolibc
          - segger
          - hal_stm32
    - name: ac6-module
      path: ac6-module
      revision: 1.0
      url: https://github.com/mymodule
```
Also check out this example https://github.com/golioth/reference-design-template/blob/main/west.yml

After setting up the west.yml file, you simply run west update to fetch the dependencies, including your custom module.
As you noted, Zephyr Workbench allows you to run west commands through the terminal, so this workflow should integrate smoothly with Zephyr Workbench's setup.

NOTE: you should not modify zephyr-v3.7.0/zephyr/west.yml, as this is T1 topology
The minimal allows the creation of a base of a minimal topoly in T2.
If you have a repo similar to https://github.com/golioth/reference-design-template with west on the topdir it works, just provide the correct tag.

zafersn, 2024-10-21:
Hi @tiborkiss @RoyAc6. I was going to ask a similar question, but its already created.

so, I read your discussion.

my workspace /application was created with [topology T2](https://docs.zephyrproject.org/latest/develop/west/workspaces.html#t2-star-topology-application-is-the-manifest-repository). The application is the manifest repository. @tiborkiss if you want to have a module is managed by a Git repository I think you can follow up this repo https://github.com/zephyrproject-rtos/example-application. But anyway, my question is here that I cant see my custom board during importing an existing project.

RoyAc6, 2024-10-21:
Ok, I see the issue now. There is a bug in how we search for and detect boards on application creation and "change board". We are working on a complete fix, but we are facing some challenges.

As a workaround, you can select any random board during application creation, and then manually change the board in the .vscode/settings.json file as follows:

```
"zephyr-workbench.board": "your_board",
"C_Cpp.default.compileCommands": "${workspaceFolder}\\build\\your_board\\compile_commands.json",

```

Would this fix it ?

zafersn, 2024-10-21:
> ```
> "C_Cpp.default.compileCommands": "${workspaceFolder}\\build\\your_board\\compile_commands.json",
> ```

No, not really.  I changed the line in the workspace

[image]

then I tried the build. but still it is not building with my board config.
And Im still seeing the defualt board, which I selected it at creating it
[image]

RoyAc6, 2024-10-21:
I don’t understand why it’s not changing. Theoretically, you should be able to put anything in
`"zephyr-workbench.board": "toto_titi_tata"`
[image]
And it should still work! Did you save the `settings.json` file? Click also on the refresh button
I'm concerned there might be a conflict with other extensions or a VSCode cache issue.
Could you try installing the[portable version](https://code.visualstudio.com/docs/editor/portable) of VSCode (the .zip version), then create the data folder and reinstall the host tools?

tiborkiss, 2024-10-22:
Continuing with the custom board and west.yml pulled from git repo..

With the recommended west.yml coming from git repo, I can initialize and build with west including the custom board.
(of course, beside west.yml I have a src/main.c and a small hello world like app.)
Here is the west.yml

```
yaml
manifest:
  version: 0.8

  remotes:
    - name: zephyrproject
      url-base: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
      repo-path: zephyr
      remote: zephyrproject
      revision: v3.7.0
      import:
        path-prefix: deps
        name-allowlist:
          - cmsis
          - cmsis-dsp
          - percepio
          - picolibc
          - segger
          - hal_stm32
    - name: xyz-zephyr-boards
      path: xyz-zephyr-boards
      revision: main
      url: https://git.../xyz-zephyr-boards

  self:
    path: app
```

Then when I want to load in Zephyr Workbench, the zephyr workspace is recognized, but the Applications view is empyt. I see just the Create New and Import existing application. Unfortunately none of the two (import existing application and create new application) works anymore in this workspace.
Just to summarize, I can begin creating the project and import the custom board and use it in newly created applications within VSCode and Zephyr Workbench. I can also initialize by west.yml using west command line tool, using a custom board board pulled from github. But these two approaches doesn't intersect somewhere to be able to continue working using either VSCode or just command line west.

RoyAc6, 2024-10-22:
We are working on an update that is going to find the boards using west, the only issue that west needs to generate some files, we are trying to optimize this step.

tiborkiss, 2024-10-23:
When trying to import existing application, just pulled from git repo by west command line tool, creates this exception
```
2024-10-23 13:57:23.300 [error] TypeError: Cannot read properties of undefined (reading 'replace')
	at Ig.file (file:///c:/Users/Tibor/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:30:58326)
	at ci (c:\Users\Tibor\.vscode\extensions\ac6.zephyr-workbench-1.1.2\out\extension.js:157:14946)
	at xr (c:\Users\Tibor\.vscode\extensions\ac6.zephyr-workbench-1.1.2\out\extension.js:157:15098)
	at c:\Users\Tibor\.vscode\extensions\ac6.zephyr-workbench-1.1.2\out\extension.js:690:6375
	at T0.h (file:///c:/Users/Tibor/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:114:32845)
	at T0.g (file:///c:/Users/Tibor/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:114:31919)
	at T0.executeCommand (file:///c:/Users/Tibor/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:114:31826)
	at Object.executeCommand (file:///c:/Users/Tibor/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:156:37049)
	at Zf (c:\Users\Tibor\.vscode\extensions\ac6.zephyr-workbench-1.1.2\out\extension.js:163:7184)
```

tiborkiss, 2024-10-25:
> A new version (1.2.0) was published, and i hope it solves your issues.
> 
> The board lists are now discovered using "west boards" command. It also searches under the west workspace folder by default. Additionnally, it supports several Zephyr environment variables (BOARD_ROOT under west workspace) and (EXTRA_ZEPHYR_MODULES under application).

I checked with version (1.2.0). Indeed now the custom board is detected from my west manifest initialized from git. I still have the custom board as external module.

In the other hand I can only generate new application, I cannot import an existing one. For this I will open another isse and I close this issue as resolved.

closed 2024-10-25 by tiborkiss