Ac6Embedded/vscode-zephyr-workbench #5: When importing existing app, the lack of some json files from .vscode prohibit importing the app.

tiborkiss, 2024-10-25:
Using v1.2.0 of Zephyr workbench when I want to import an existing application (using the button from APPLICATIONS view), that fails if the application was not created by Zephyr workbench before.

I noticed that when Zephyr workbench initializes an application, adds certain json files in .vscode folder.
If I create an application using Zephyr Workbench, then I remove and add again. The import works correctly.
I did this test because my application which I uploaded to my git repo, containing also the west manifest file, when I initialize first using command line `west init -m <my-git-repo-url-to-the-app-with-west-manifest> .` I need to import into Zephyr workbench.
When I try to import, seemingly pass also the custom board discovery, but the import fails with the following error.

```
2024-10-25 10:45:35.733 [error] TypeError: Cannot read properties of undefined (reading 'identifier')
	at xa (c:\Users\Tibor\.vscode\extensions\ac6.zephyr-workbench-1.2.0\out\extension.js:161:3905)
	at async c:\Users\Tibor\.vscode\extensions\ac6.zephyr-workbench-1.2.0\out\extension.js:701:6397
	at async T0.h (file:///c:/Users/Tibor/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:114:32837) zephyr-workbench-app-explorer.import-app {"value":"Ac6.zephyr-workbench","_lower":"ac6.zephyr-workbench"}
2024-10-25 10:45:36.748 [error] TypeError: Cannot read properties of undefined (reading 'identifier')
	at xa (c:\Users\Tibor\.vscode\extensions\ac6.zephyr-workbench-1.2.0\out\extension.js:161:3905)
	at async c:\Users\Tibor\.vscode\extensions\ac6.zephyr-workbench-1.2.0\out\extension.js:701:6397
	at async T0.h (file:///c:/Users/Tibor/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/api/node/extensionHostProcess.js:114:32837)
```
I simplified my manifest file, to not use any custom board, but initializing during import with an official `samd21_xpro`. The exact same error I see in the Extensions Host logs.

Since my application I did not created using Zephyr workbench, lacks certain json files from .vscode folder.
I took a hello_world example which was working well also with import existing application, then I removed the json files from .vscode folder and I could reproduce the same errors as with my app initialized with west manifest from private repo.

Then I took those json files and I added to my app which resulted in successful importing by Zephyr workbench.

Therefore I concluded that those json files in .vscode folder, generated by Zephyr Workbench during application creation, are simply missing during importing of applications. I think that the fix is very straighforward, the initialization of the json files have to be added to the application import if they are not present. Or, better updating the existing ones, by adding the required keys and attributes.

Until that fix is made. The workaround is simple, we save those json files from an existing app create by Zephyr Workbench and manually we add to our project which we would like to import but don't have these.

As as side note. The Applications view lacks the possibility to remove an application without deleting from disk. During my test, I always had to save the app folder first somewhere, then use "Delete from disk" menu and manually copying back the folder to import again. I think it would be useful to be able to remove the application without deleting from disk.

KevinTangDev, 2024-10-25:
Can you confirm me that in your boardname.yaml file, there is an "identifier" key with a value ? The error happens in the function that generates the values into a new or existing _.vscode/settings.json_ after parsing the yaml file.

About removing an application without deleting from disk, you're right, i will be usefull. The alternative is to "Remove the folder from workspace" in the Explorer view.

[image]

tiborkiss, 2024-10-25:
Regarding removing the folder... my bad. I removed also from there but I wrongly remembered that the application still remain in the list. That was wrong observation.
In this way, for me is sufficient to remove from Explorer view.

Tha board yaml file I used is exactly this https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/atmel/sam0/samd21_xpro/samd21_xpro.yaml
I copied samd21_xpro and replaced the samd21_xpro values and all the renamings which usually have to do when making custom boards from an already existing ones.

I confirm that there is a value associated to "identifier" key.

Even more, you don't need to create custom board, just use this samd21_xpro  on a hello_world app.
First time you create the hello_world app, and use samd21_xpro board. Then remove the app (like you showed in the picture above). In addition, delete the content of .vscode from hello_world folder and than try to import it again using samd21_xpro board. In case I removed the json files from .vscode, I get that exception, mentioning the "identifier".

KevinTangDev, 2024-10-25:
I found out the issue (on the import ui, it was searching board from name instead of identifier). It is fixed on the new update.

tiborkiss, 2024-10-25:
Excellent. I am already busy with my other things, only Monday I will be able to test at this time.

After my tests I concluded that how it is now the Zephyr Workbench, it is a great tool.
The most I like the fact that non-intrusively follows the west build given structures and recipes.

tiborkiss, 2024-10-28:
I tested and it works. Thank you for the quick fix.

closed 2024-10-28 by tiborkiss