Ac6Embedded/vscode-zephyr-workbench #6: Often debug fails to start because missing west_wrapper.bat

tiborkiss, 2024-11-04:
Using v1.2.4 of the plugin I noticed the following behavior:
After I built or pristine built a project I see that in build/<board>/ folder I see the `west_wrapper.bat`. Then I call the Debug Manager, configure the gdb and debugserver (currently using openocd). When I want to debug, starts to rebuild everything and at the end is missing west_wrapper.bat. It fails in the moment when the cortexdebug is popping up.
The console output looks like this.

```
1: (70) LaunchOptions{"name":"Zephyr Workbench Debug","type":"cppdbg","request":"launch","cwd":"E:\\w\\easy6\\easy-ws\\wired_uart_out","preLaunchTask":"West Build","program":"E:\\w\\easy6\\easy-ws\\wired_uart_out\\build\\samd21_xpro\\zephyr\\zephyr.elf","args":[],"stopAtEntry":true,"svdPath":"e:\\w\\easy6\\svd\\ATSAMD21J18A.svd","environment":[],"externalConsole":false,"serverLaunchTimeout":10000,"filterStderr":true,"filterStdout":true,"serverStarted":"halted due to debug-request, current mode: Thread","MIMode":"gdb","miDebuggerPath":"c:\\Users\\Tibor\\zephyr-sdk-0.16.8\\arm-zephyr-eabi\\bin\\arm-zephyr-eabi-gdb.exe","debugServerPath":"E:\\w\\easy6\\easy-ws\\wired_uart_out/build/samd21_xpro/west_wrapper.bat","debugServerArgs":"debugserver --build-dir E:\\w\\easy6\\easy-ws\\wired_uart_out/build/samd21_xpro --runner openocd --config openocd.cfg --config E:\\w\\easy6\\easy-ws\\wired_uart_out/build/samd21_xpro/gdb.cfg ","setupCommands":[{"text":"-environment-cd E:\\w\\easy6\\easy-ws\\wired_uart_out\\\\build\\\\samd21_xpro\\\\zephyr"},{"text":"-target-select remote localhost:3333","description":"connect to target","ignoreFailures":false},{"text":"-file-exec-and-symbols zephyr.elf","description":"load file","ignoreFailures":false},{"text":"-interpreter-exec console \"monitor reset\"","ignoreFailures":false},{"text":"-target-download","description":"flash target","ignoreFailures":false},{"text":"set breakpoint pending on","description":"Set pending","ignoreFailures":false},{"text":"tbreak main","description":"Set a breakpoint at main","ignoreFailures":true}],"logging":{"moduleLoad":true,"trace":true,"engineLogging":true,"programOutput":true,"exceptions":true},"__configurationTarget":6,"configSource":"workspaceFolder","debugType":"debug","__sessionId":"e03d7295-50ee-47cb-acc6-ccfac3917559"}
1: (115) Starting: "E:\w\easy6\easy-ws\wired_uart_out/build/samd21_xpro/west_wrapper.bat" debugserver --build-dir E:\w\easy6\easy-ws\wired_uart_out/build/samd21_xpro --runner openocd --config openocd.cfg --config E:\w\easy6\easy-ws\wired_uart_out/build/samd21_xpro/gdb.cfg 
```

This `wired_uart_out` app basically is a copy of the hello_world.
I tried multiple combinations. For example, when I create new app from hello_world sample, when was just created I can debug. Then I clean the build directory, change the board, built with that, than clean again, switch back to the samd21_xpro board, and than build and configure the Debug Manager again and fails.
Then I just copied the original hello_world and renamed to `wired_uart_out`. That fails from the first attempt.
I tried to delete build folder, including .vscode folder and redo. Still the same.
But sometimes, not so often still works.

KevinTangDev, 2024-11-04:
The `west_wrapper.bat` is only generated in the build directory when you click on Apply or Debug from the Debug Manager. If the project requires pristine built, it will remove items from the build directory and the wrapper as well... We may need to generate the wrapper out of the build directory to avoid that or find a way to generate the file after rebuild.

tiborkiss, 2024-11-05:
I already noticed that build will delete the wrapper, therefore I always applied the debug manager configuration.
I recorded a video how it happens. You will see two issues in the video.
1 - If an app sits in the workspace folder and has no build folder, nor .vscode folder, when I import the app, I have to do it twice.
2 - The wrapper is deleted while in Debug Manager I press Apply and Debug. Apply is creating the wrapper, then debug somehow is triggering the rebuild, which deletes the wrapper.

[the recorded video is on youtube, by clicking here](https://youtu.be/lSh0GTOf3SY)

Note, that the video was recorded by OBS with window capture feature, therefore the popup windows during import and the final error window are not visible. The final error is in the console and is trying to tell us that the debug wrapper is missing, which is true.

tiborkiss, 2024-11-05:
Continuing...
Now I compared the content between the application folder which fails and which doesn't fail to debug.
In `.vscode/compile_commands.json` I see a difference. The version which works still points to zephyr/samples/hello_world folder, despite I created the application with folder location filled to be in the top folder of my workspace. Means that, the copy of the app was made into the top folder, but the compile_commands.json still points back to the original place from where it was copied.
Then when I delete the .vscode folder, this compile_commands.json which has these paths to the original location is deleted. When I re-import the app, the compile_commands.json is recreated with the correct content. Unfortunately, the Debug Monitor, Apply and Debug in this case somehow triggers the rebuild, which zaps the debug wrapper.bat

To be able to fix, first one must fix the application creation process, to not sneak in the original path from where it was copied. Then, I expect that the Debug Manager, Apply and Debug steps uniformely will fail with any of your applications. Similarily when we just delete .vcode from the app and we would like to re-import again.

KevinTangDev, 2024-11-05:
> I already noticed that build will delete the wrapper, therefore I always applied the debug manager configuration. I recorded a video to show you how it happens. You will see two issues in the video. 
1 - If an app sits in the workspace folder and has no build folder, nor .vscode folder, when I import the app, I have to do it twice. 

I find out that the function to check if the workspace folder is an app (by checking tasks) is called before VS Code had finished parsing newly created `.vscode` which causes the missing app in the view. The solution is to refresh the view only after `.vscode/tasks.json` changes are detected by VS Code.

>2 - The wrapper is deleted while in Debug Manager I press Apply and Debug. Apply is creating the wrapper, then debug somehow is triggering the rebuild, which deletes the wrapper.

The wrapper will be generated in `<app>/build/.debug/<board_identifier>` instead of `<app>/build/<board_identifier>`. The rebuild won't affect the files generated by Zephyr Workbench anymore.

> 
> [the recorded video is on youtube, by clicking here](https://youtu.be/lSh0GTOf3SY)
> 
> Note, that the video was recorded by OBS with window capture feature, therefore the popup windows during import and the final error window are not visible. The final error is in the console and is trying to tell us that the debug wrapper is missing, which is true.
> 
> One more note: If I import the same hello_world app, by leaving in the original zephyr/samples folder, the above issue doesn't happen. To reproduce, you have two options (practically the same):
> 
> * You must copy the hello_world folder into the top folder of the workspace,
> 
> * Or an already installed hello_world app (using the Create New Zephyr Application button, and specifying the project location to be the top folder), then removing from workspace and deleting build and .vscode folder. Similarily with situations when we first create or copy a clean app.

closed 2024-11-05 by KevinTangDev

KevinTangDev, 2024-11-05:
The fix is published in the version 1.2.5

commit 951269f0 (2024-11-05): Version 1.2.5

Fix #6
Application auto-refresh after tasks.json modified
Generating west wrapper and other debug files into `build/.debug/<board_identifier>`