Ac6Embedded/vscode-zephyr-workbench #11: Multiple builds for same project

rokobertcm, 2024-11-19:
Is it possible to implement multiple builds for one project in multiple build directories.
Reason why I think that we needed is that you could then have for example:
One build for debugging purpose. (different .prj files)
One build for release (different prj. files.
One build for release with west --sysbuild option.

Regards,
Robert

KevinTangDev, 2024-11-20:
Multiple builds is supported yet. Thank you for the suggestion. I will work on this feature asap.

KevinTangDev, 2025-02-03:
The multiple builds feature has been release a month ago and currently, the feedback are positive. 

About usage, please refer to documentation:
How to configure application build: https://zephyr-workbench.com/docs/documentation/application
How to add multibuild: https://zephyr-workbench.com/docs/documentation/multibuild

closed 2025-02-03 by KevinTangDev