Ac6Embedded/vscode-zephyr-workbench #12: custom BOARD_ROOT path

sebistark, 2024-12-19:
What is the correct syntax for adding a custom board path?
I have tried it with CLI interface and west and there my board is recognised

Note: the boards directory is a parent directory of the zephyr directory 
[image]

RoyAc6, 2024-12-19:
Hi @sebistark chech out this discusson #4 for best practices
you must specify the path of a folder that contains a "boards" directory, preferably with absolute path

sebistark, 2024-12-20:
Hi @RoyAc6 
Thank you very much for your answer. Indeed, I had to specify the path to the parent directory of "boards".
In my case it was "my_boards"

my_boards/
└── boards/
    └── custom_vendor/
        ├── board_name1/
        │   ├── board_name1.dts
        │   ├── board_name1.yaml
        │   └── ...
        └── board_name2/
            ├── board_name2.dts
            ├── board_name2.yaml
            └── ...
.
.
.
my_app/
├── src/
│   └── main.c
├── CMakeLists.txt
├── prj.conf
└── ...

closed 2024-12-20 by sebistark