Ac6Embedded/vscode-zephyr-workbench #13: Can not create/import projects (on devcontainer)

jonppe, 2025-02-02:
I tried to use the plugin on devcontainer.
Haven't yet tried without container so I'm not quite sure if that is the reason for issues.

Anyway, when using the "Create New Application" the "Select Board" and "Select Sample project" dropdowns don't seem to get any content

[image]

When using the "Import exising Application", there is a similar endless wait with a title "Updating available boards list"

It would be nice to get this working inside a container to make it easier to share (exact descriptions of) dev-environments with other devs that are a bit safer too.

RoyAc6, 2025-02-02:
Hi @jonppe, it appears the issue is caused by using space characters in the name of the west workspace. To prevent this problem, please ensure that no spaces are used in the workspace path or the SDK name.
Could you please confirm that everything works correctly in a container using a path without special characters?
We will work on a fix
Thanks

KevinTangDev, 2025-02-03:
There is indeed an issue with the empty space in the west workspace name. I just fixed it. An update will be publish today.
About using the extension within devcontainer, I did not test it yet. I will check if there is any other compatibility issues.

jonppe, 2025-02-04:
I tried this again today and realized that there actually wasn't space in the name but underscore _.
Behavior seems the same with the latest version so I'm guessing this could be container related.
I'm not very familiar on VSCode extensions but I briefly decided to look in the dev-tools console:

There were some "Timed out getting tasks..." errors when opening Zephyr Workspace tab.
Then some security policy issues when clicking "Create New Application".
And finally some last "Error" I selected the project

[image]

KevinTangDev, 2025-02-05:
Could you tell me which docker base image you use ? I just tested with the following Dockerfile (with a ubuntu image), and i couldn't reproduce the issue.

```
FROM ubuntu:latest

# Set the maintainer label
LABEL maintainer="kevin.tang@ac6.fr"

# Update package list and install the necessary packages
RUN apt-get update && \
    apt-get install -y \
    build-essential \
    wget \
    sudo \
    pkexec \
    unzip \
    gperf \
    git \
    ccache \
    dfu-util \
    xz-utils \
    file \
    make \
    libsdl2-dev \
    libmagic1 \
    python3 \
    openssl  
    
# Clean up apt cache to reduce image size
RUN rm -rf /var/lib/apt/lists/*

RUN mkdir -p /root/workdir/sdk /root/workdir/zephyr /root/workdir/apps

CMD ["/bin/bash"]

```

closed 2025-10-17 by RoyAc6