Ac6Embedded/vscode-zephyr-workbench #15: Working with an out of tree board failed !

navr32, 2025-04-02:
Hi ! Many thanks for this very promising vscode extension for zephyr.

I just come to try it. Before I was working with a manually created out  of tree board on the cli.

So i was thinking this is easy to use my out of tree board with Zephyr Workbench and zephyr 4.1 and sdk 0.17

So i  have first let Zephyr Workbench generate a new West workspace and after this just create a "New Application" blinky
and select a generic board because i haven't see any option to import an existing out of tree board  from here ! (perhaps i have missing something ?).

So the new blinky was created i test to build it with the dev board selected and  all build ok and flash ok to my dev board . Very happy.

After this i want to use my custom board so i have add manually to the blinky app directory my boards directory with my own vendor/boardmodels with all it's own file yaml..dts..and so....

And when all  of this files is in good directory I  click right on the blinky  application the menu come on ---> configure --> change board..

The list of change board come on on the center top panel and  i can view all the boards zephyrs or west can see installed ok..
But my out of tree board never show..so i have try many change off settings.. but unable to have my board come on in this list
to select it.

I have close re open vscode..the extension..nothing....i have try  to force the west workspace "board root " path with the path of the blinky app..without  "boards/vendos/modelboards" ...and no card detected...i have try to set my boards directory in the west workspace root...and test and again to set the "board root" path ..to the same path of the west workbench for this case again no board detection ?

So i have seen Nordic semiconductor have done some plugin for zephyrs i insalled them and try and the nordic plugin list my out of tree board if i put the board in the blinky app i created with him and the zephyr plugin let me  select my custom boards if i try to change the board to build !

RoyAc6, 2025-04-02:
Hi @navr32, thank you for your feedback.
Custom boards should work fine, it should list anything detected using `west boards`
Where did you define your custom board ? is it in an OoT module, with a module.yml that contains `board_root: .` ?
Did you set the custom board using BOARD_ROOT in the west workspace settings ? https://zephyr-workbench.com/docs/documentation/custom ?

[image]

You can always force the option by modifying your .vscode/settings.json : `"zephyr-workbench.board": "my_board"`

RoyAc6, 2025-04-02:
you can also check #12 and #4

RoyAc6, 2025-04-05:
We have detected the source of the error, we are working on a fix, it will fixed in the next release, beginning of next week.

SamDCR, 2025-04-07:
Hi @navr32,

The new version of the extension is now available. It includes the fix for the out-of-tree board detection issue you mentioned.

When you have a chance, please test it and let us know if your custom board is now listed as expected. If you still encounter any issues, feel free to reach out—we’ll be available to help.

Thanks again for your feedback.

dolence, 2025-06-18:
Custom board is showing when creating a new project but not when importing an existing project. Any thoughts?