Ac6Embedded/vscode-zephyr-workbench #16: build folder is confuse which prevent west to build/flash/debug

mconstant68, 2025-04-04:
HI I love the addon, hopefully it will be possible to fix this issue.

I am using the latest of VSCODEV1.98.2

It seems that evrything seems to work properly except that I have these 3 file/folders that gets created randomly outside of the `{path...}\build\primary`.

1. `CMakeCache.txt`
2. `CMakeFiles`
3. `KConfig`

It is not easy to reproduce. It is a bit random. I have monitored the folder to see when they appear, and it is random.

Here is a screenshot of my vscode. This is really anoying. Hopefully you can reproduce it.

[image]

RoyAc6, 2025-04-05:
Hi @mconstant68 , thank you!

Regarding the issue, I'm not entirely sure if it's related to `west build/flash`. However, the files being created seem to be due to another extension. From your screenshot, I noticed that you have the STM32 extension installed, which depends on [twxs.cmake](https://marketplace.visualstudio.com/items?itemName=twxs.cmake). This extension automatically generates those files whenever it detects a CMakeLists.txt.

This behavior is generally not a problem, but it can "pollute" the project a bit with extra files.

mconstant68, 2025-04-07:
This is interesting. I will try this out. Does the `STM32` and `twxs.cmake` addons are a dependency for Zephyr Workbench?

closed 2025-04-08 by mconstant68