Ac6Embedded/vscode-zephyr-workbench #17: Unable to debug a stm32h735g_disco board using the Launch.json configuration

mconstant68, 2025-04-08:
The problem seems to be related to the `launch.json` configuration file but I can't seem to see what the problem is.

[launch.json](https://github.com/user-attachments/files/19653413/launch.json)

Here is the problem. I created a new application using the stm32h735g_disco board with the `sample/basec/blinky` application.

The project compile properly using the build button, then I am able to flash the application with either `STM32CubeProgrammer` or `STLink(openocd)`.

I create a debug entry with the `Debug Manager` (see below screen shot).

[image]

The issue happens when I try to debug, the debugger is not able to start properly. The `DEBUG CONSOLE` shows:

```
1: (75) LaunchOptions{"name":"Zephyr Workbench Debug [primary]","type":"cppdbg","request":"launch","cwd":"C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g","program":"C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g\\build\\primary\\zephyr\\zephyr.elf","args":[],"stopAtEntry":true,"svdPath":"","environment":[],"externalConsole":false,"serverLaunchTimeout":30000,"filterStderr":true,"filterStdout":true,"serverStarted":"halted due to debug-request, current mode: Thread","MIMode":"gdb","miDebuggerPath":"c:\\Zephyr\\SDK_4_1_0\\zephyr-sdk-0.17.0\\arm-zephyr-eabi\\bin\\arm-zephyr-eabi-gdb.exe","debugServerPath":"C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g\\build\\primary\\.debug\\west_wrapper.bat","debugServerArgs":"debugserver --build-dir \"C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g/build\\primary\" --runner openocd --gdb-port 3333 --openocd C:\\Users\\mconstant\\.zinstaller\\tools\\openocd\\bin\\openocd.exe --config openocd.cfg --config C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g/build/.debug/gdb.cfg ","setupCommands":[{"text":"-environment-cd C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g\\\\build\\\\primary\\\\zephyr"},{"text":"-target-select remote localhost:3333","description":"connect to target","ignoreFailures":false},{"text":"-file-exec-and-symbols zephyr.elf","description":"load file","ignoreFailures":false},{"text":"-interpreter-exec console \"monitor reset\"","ignoreFailures":false},{"text":"-target-download","description":"flash target","ignoreFailures":false},{"text":"set breakpoint pending on","description":"Set pending","ignoreFailures":false},{"text":"tbreak main","description":"Set a breakpoint at main","ignoreFailures":true}],"logging":{"moduleLoad":true,"trace":true,"engineLogging":true,"programOutput":true,"exceptions":true},"__configurationTarget":6,"__sessionId":"a458a7ce-ae32-49a7-a6e1-17a13b0ab930"}
1: (113) Starting: "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g\build\primary\.debug\west_wrapper.bat" debugserver --build-dir "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build\primary" --runner openocd --gdb-port 3333 --openocd C:\Users\mconstant\.zinstaller\tools\openocd\bin\openocd.exe --config openocd.cfg --config C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/.debug/gdb.cfg 
1: (260) ->Activated virtual environment at "C:\Users\mconstant\.zinstaller\.venv\Scripts\activate.bat"
1: (1067) ->-- west debugserver: rebuilding
1: (1236) ->ninja: no work to do.
1: (1248) ->-- west debugserver: using runner openocd
1: (1250) ->-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available
1: (1347) "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g\build\primary\.debug\west_wrapper.bat" exited with code 1 (0x1).
1: (1350) <-logout
1: (30154) <-logout
1: (30155) Send Event AD7MessageEvent
```

And 30 seconds or so after, I get this dialog box.

[image]

However, if I open a west terminal and run

`west debug  --build-dir "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/primary"`

I am able to debug step by step. (see below)

```
(.venv) PS C:\Users\mconstant\cirrus_vscode> west debug  --build-dir "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/primary"
-- west debug: rebuilding
ninja: no work to do.
-- west debug: using runner openocd
-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available
GNU gdb (Zephyr SDK 0.17.0) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later 
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=arm-zephyr-eabi".
Type "show configuration" for configuration details.
_
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from C:/Users/mconstant/cirrus_vscode/projects/blinky_h735g/blinky_h735g/build/primary/zephyr/zephyr.elf...
Remote debugging using :3333
z_arm_reset () at C:/Users/mconstant/cirrus_vscode/deps/zephyr/arch/arm/core/cortex_m\reset.S:124
124         movs.n r0, #_EXC_IRQ_DEFAULT_PRIO
Loading section rom_start, size 0x2cc lma 0x8000000
Loading section text, size 0x8834 lma 0x80002d0
Loading section .ARM.exidx, size 0x8 lma 0x8008b04
Loading section initlevel, size 0xa8 lma 0x8008b0c
Loading section device_area, size 0x168 lma 0x8008bb4
Loading section sw_isr_table, size 0x518 lma 0x8008d1c
Loading section gpio_driver_api_area, size 0x24 lma 0x8009234
Loading section reset_driver_api_area, size 0x10 lma 0x8009258
Loading section clock_control_driver_api_area, size 0x1c lma 0x8009268
Loading section uart_driver_api_area, size 0x4c lma 0x8009284
Loading section shell_area, size 0x2c lma 0x80092d0
Loading section shell_root_cmds_area, size 0x34 lma 0x80092fc
Loading section shell_subcmds_area, size 0x118 lma 0x8009330
Loading section rodata, size 0x1c1c lma 0x8009448
Loading section datas, size 0xc4 lma 0x800b064
--Type <RET> for more, q to quit, c to continue without paging--b main
Loading section device_states, size 0x1e lma 0x800b128
Loading section .last_section, size 0x4 lma 0x800b146
Start address 0x08002ec8, load size 45382
Transfer rate: 6 KB/sec, 2388 bytes/write.
(gdb) b main
Breakpoint 1 at 0x80007b0: file C:/Users/mconstant/cirrus_vscode/projects/blinky_h735g/blinky_h735g/build/primary/zephyr/include/generated/zephyr/syscalls/device.h, line 57.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) c
Continuing.
_
Breakpoint 1, main () at C:/Users/mconstant/cirrus_vscode/projects/blinky_h735g/blinky_h735g/build/primary/zephyr/include/generated/zephyr/syscalls/device.h:57
57              compiler_barrier();
(gdb)
```

Any ideas?

mconstant68, 2025-04-08:
It seems that it does not know `debugserver`

```
(.venv) PS C:\Users\mconstant\cirrus_vscode> "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g\build\primary\.debug\west_wrapper.bat" debugserver --build-dir "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build\primary" --runner openocd --gdb-port 3333 --openocd C:\Users\mconstant\.zinstaller\tools\openocd\bin\openocd.exe --config openocd.cfg --config C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/.debug/gdb.cfg
At line:1 char:109
+ ... inky_h735g\build\primary\.debug\west_wrapper.bat" debugserver --build ...
+                                                       ~~~~~~~~~~~
Unexpected token 'debugserver' in expression or statement.
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : UnexpectedToken

(.venv) PS C:\Users\mconstant\cirrus_vscode> 
```

RoyAc6, 2025-04-08:
Sorry, you need to remove the double quotes, try this:
`C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g\build\primary\.debug\west_wrapper.bat debugserver --build-dir "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build\primary" --runner openocd --gdb-port 3333 --openocd C:\Users\mconstant\.zinstaller\tools\openocd\bin\openocd.exe --config openocd.cfg --config C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/.debug/gdb.cfg `

mconstant68, 2025-04-08:
Here is the results:

```
Activated virtual environment at "C:\Users\mconstant\.zinstaller\.venv\Scripts\activate.bat"
-- west debugserver: rebuilding
ninja: no work to do.
-- west debugserver: using runner openocd
-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available
Open On-Chip Debugger 0.11.0+dev-00728-gb6f95a16c (2024-12-04-15:09)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
init
C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/.debug/gdb.cfg:3: Error: can't read "_TARGETNAME": no such variable
in procedure 'script'
at file "embedded:startup.tcl", line 26
at file "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/.debug/gdb.cfg", line 3
FATAL ERROR: command exited with status 1: 'C:\Users\mconstant\.zinstaller\tools\openocd\bin\openocd.exe' -s 'C:/Users/mconstant/cirrus_vscode/deps/zephyr/boards/st/stm32h735g_disco\support' -s 'C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/.debug' -s 'C:\Program Files/OpenOCD/share/openocd/scripts' -f openocd.cfg -f 'C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/.debug/gdb.cfg' -c 'tcl_port 6333' -c 'telnet_port 4444' -c 'gdb_port 3333' '-c init' '-c targets' -c 'reset init'
(.venv) PS C:\Users\mconstant\cirrus_vscode>
```

I also tried to delete the lauch.json and create a new debug configuration and I got the same issue.

mconstant68, 2025-04-08:
In case you wonder, I have installed all the tools you requested.

[image]

But it looks like it is using another version of OpenOCD's scripts installed in `"C:\Program Files/OpenOCD/share/openocd/scripts"`

RoyAc6, 2025-04-08:
Can you try to use the other OpenOCD installation that is under `C:/Program Files/OpenOCD`, you can set it explicitly under Debug Manager, in Runner Path, you should specify openocd.exe

Is it only with this board ? or you had the same issue with other boards ? 
We use this fork of OpenOCD it is recommended by Zephyr but it is not the latest version, https://github.com/zephyrproject-rtos/openocd

mconstant68, 2025-04-08:
I am getting the same error.

```
1: (71) LaunchOptions{"name":"Zephyr Workbench Debug [primary]","type":"cppdbg","request":"launch","cwd":"C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g","program":"C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g\\build\\primary\\zephyr\\zephyr.elf","args":[],"stopAtEntry":true,"svdPath":"","environment":[],"externalConsole":false,"serverLaunchTimeout":30000,"filterStderr":true,"filterStdout":true,"serverStarted":"halted due to debug-request, current mode: Thread","MIMode":"gdb","miDebuggerPath":"c:\\Zephyr\\SDK_4_1_0\\zephyr-sdk-0.17.0\\arm-zephyr-eabi\\bin\\arm-zephyr-eabi-gdb.exe","debugServerPath":"C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g\\build\\primary\\.debug\\west_wrapper.bat","debugServerArgs":"debugserver --build-dir \"C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g/build\\primary\" --runner openocd --gdb-port 3333 --openocd c:\\Program Files\\OpenOCD\\bin\\openocd.exe --config openocd.cfg --config C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g/build/.debug/gdb.cfg ","setupCommands":[{"text":"-environment-cd C:\\Users\\mconstant\\cirrus_vscode\\projects\\blinky_h735g\\blinky_h735g\\\\build\\\\primary\\\\zephyr"},{"text":"-target-select remote localhost:3333","description":"connect to target","ignoreFailures":false},{"text":"-file-exec-and-symbols zephyr.elf","description":"load file","ignoreFailures":false},{"text":"-interpreter-exec console \"monitor reset\"","ignoreFailures":false},{"text":"-target-download","description":"flash target","ignoreFailures":false},{"text":"set breakpoint pending on","description":"Set pending","ignoreFailures":false},{"text":"tbreak main","description":"Set a breakpoint at main","ignoreFailures":true}],"logging":{"moduleLoad":true,"trace":true,"engineLogging":true,"programOutput":true,"exceptions":true},"__configurationTarget":6,"__sessionId":"fb961a1f-06df-4d46-8171-e5c36aaf77f4"}
1: (105) Starting: "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g\build\primary\.debug\west_wrapper.bat" debugserver --build-dir "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build\primary" --runner openocd --gdb-port 3333 --openocd c:\Program Files\OpenOCD\bin\openocd.exe --config openocd.cfg --config C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/.debug/gdb.cfg 
1: (224) ->Activated virtual environment at "C:\Users\mconstant\.zinstaller\.venv\Scripts\activate.bat"
1: (1501) ->-- west debugserver: rebuilding
1: (1645) ->ninja: no work to do.
1: (1664) ->-- west debugserver: using runner openocd
1: (1707) "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g\build\primary\.debug\west_wrapper.bat" exited with code 1 (0x1).
1: (1710) <-logout
1: (30136) <-logout
1: (30138) Send Event AD7MessageEvent
```

mconstant68, 2025-04-08:
I forgot to mention that when I use the b_g474e_pow1 board, I do not have this issue.

RoyAc6, 2025-04-08:
The detailled log, you can get is using the command that is extracted from the Debug Console:
`C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g\build\primary\.debug\west_wrapper.bat debugserver --build-dir "C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build\primary" --runner openocd --gdb-port 3333 --openocd c:\Program Files\OpenOCD\bin\openocd.exe --config openocd.cfg --config C:\Users\mconstant\cirrus_vscode\projects\blinky_h735g\blinky_h735g/build/.debug/gdb.cfg `

you can try the command without the 
`--config .../build/.debug/gdb.cfg `

it seems like the _TARGETNAME does not exist for some reason.

mconstant68, 2025-04-09:
I did try without the `--config .../build/.debug/gdb.cfg`. But it is still not working. I have tried few thing but this is not easy. I am navigating thru something completely new for me. I am not sure how I can help. I am certainly confused by the fact that if I use `west debug -r openocd`, it works but using the `Zephyr Workbench` there are some issues. Not sure why. I can also confirm that using the 'b_g474e_pow1' board with the same 'blinky' sample app, it works like a charm. :-(

mconstant68, 2025-04-14:
Hello @SamDCR, I am having the same issue with this new release. I tried multiple different option to try to understand where the issue is, and it seems to get down to the gbd.cfg that is not able to resolve the _CHIPNAME. Before, the _TARGETNAME was not resolved, now it is the _CHIPNAME.

```
# Workaround to force OpenOCD to shutdown when gdb is detached (auto-generated)

    if {[info exists _TARGETNAME]} {
      $_TARGETNAME configure -event gdb-detach {
        shutdown
      }
    } else {
      # Fallback targets using CHIPNAME-based naming
      if {[info exists _CHIPNAME]} {
        if {[target names] contains "$_CHIPNAME.cpu0"} {
          $_CHIPNAME.cpu0 configure -event gdb-detach {
            shutdown
          }
        }
        if {[target names] contains "$_CHIPNAME.cpu1"} {
          $_CHIPNAME.cpu1 configure -event gdb-detach {
            shutdown
          }
        }
      }
    }
```

SamDCR, 2025-04-16:
Hi @mconstant68,

Thanks again for your thorough testing and detailed feedback — and apologies for the continued trouble.

We've just released a new version that includes an updated workaround for the issue. If you have a bit of time, could you please give it a try and let us know if it changes anything on your side?

Unfortunately, I only have access to an older (and obsolete) STM32H7 board (not the exact stm32h735g_disco), so I haven’t been able to fully reproduce the issue myself — which makes your feedback all the more helpful.

Your insights have been really valuable, and we'd love to hear if the situation improves with this latest update — or if you're still hitting the same wall. Thanks again for your patience and support!

mconstant68, 2025-04-16:
Hi @SamDCR 

The issue is resolved. I am now able to enter debug mode without any issues and able to do step by step debugging.

Thank you so much for your help resolving this issue.

I really like your addon and hopefully it can become the Zephyr official Addon :-)

Marco

closed 2025-04-16 by mconstant68