Ac6Embedded/vscode-zephyr-workbench #18: wsl polkit issue

digiexchris, 2025-04-22:
Hey, when running the extension on a windows machine using WSL as my dev env, I get an error popup "no polkit authentication agent found" when installing the host tools. Everything appears to install, but west is failing to then correctly install or locate the espressif toolchain (missing the ESP_IDF_PATH env var during the build). Not sure if the two issues are related, so mostly just wondering if there has been any success using the extension with wsl?

RoyAc6, 2025-04-23:
Hi, thanks for the update! Could you please let us know which version you’re running? It looks like this issue may be coming from VS Code in WSL rather than Zephyr Workbench. Do you notice exactly when the error appears? According to [this post](https://stackoverflow.com/questions/72560287/no-polkit-authentication-agent-found-vs-code), you might also need to install lxpolkit to resolve it.

digiexchris, 2025-04-25:
Zephyr Workbench version 1.3.18, running on a Debian 12 guest.

Currently, since I already have it installed, It pops up the error notification as soon as I click "Yes" in the notification popup that asks me to confirm if I want to reinstall host tools, and I can't tell if it happens before or after the terminal shows the progress of the install. I have lxpolkit and polkitd installed, but that doesn't do any good because there is no x server running with a polkit session. Interestingly, if it helps, the espressif extension which does similar things works, not sure how it works around that.