Ac6Embedded/vscode-zephyr-workbench #20: Cannot setup on MacOS latest version

csann, 2025-05-11:
I have very little experience with Zephyr and my previous attempts failed due to the complexity. I am very excited to try Zephyr Workbench. Unfortunately I can't get it setup on my MacBook Pro Intel based computer. My OS is Sequoia 15.4.1.

Installing Host Tools and the Virtual Environment appear to be successful. 

**Debug Tools**

The problem starts when I install Debug Tools. 

[image]

After I use the link to the st website to install STM32CubeProgrammer, Zephyr Workbench still indicates it was not installed. The same thing happens when I install OpenOCD. The terminal indicates OpenOCD was installed but the GUI is not updated. 

```
----------------------------------------
  Parse tools definitions and generate manifest
----------------------------------------
source=https://www.ac6-tools.com/downloads/zephyr-workbench/tools/darwin/openocd-zephyr-0.11.0-darwin-aarch64.tar.xz
----------------------------------------
                openocd
----------------------------------------
INSTALLER_FILENAME=openocd-zephyr-0.11.0-darwin-aarch64.tar.xz
INFO: DL: openocd-zephyr-0.11.0-darwin-aarch64.tar.xz downloaded successfully
INFO: Extracting tar.xz archive... /Users/clark/.zinstaller/tmp/downloads/openocd-zephyr-0.11.0-darwin-aarch64.tar.xz to /Users/clark/.zinstaller/tools
x openocd/
x openocd/.DS_Store
x openocd/bin/
x openocd/bin/openocd
x openocd/share/
x openocd/share/openocd/
x openocd/share/openocd/scripts/
x openocd/share/openocd/scripts/bitsbytes.tcl
x openocd/share/openocd/scripts/mmr_helpers.tcl
x openocd/share/openocd/scripts/interface/
x openocd/share/openocd/scripts/interface/usb-jtag.cfg
x openocd/share/openocd/scripts/interface/stlink.cfg
x openocd/share/openocd/scripts/interface/xds110.cfg
x openocd/share/openocd/scripts/interface/altera-usb-blaster.cfg
x openocd/share/openocd/scripts/interface/altera-usb-blaster2.cfg
x openocd/share/openocd/scripts/interface/flashlink.cfg
x openocd/share/openocd/scripts/interface/estick.cfg
x openocd/share/openocd/scripts/interface/jtag_vpi.cfg
x openocd/share/openocd/scripts/interface/dummy.cfg
x openocd/share/openocd/scripts/interface/raspberrypi2-native.cfg
x openocd/share/openocd/scripts/interface/usbprog.cfg
x openocd/share/openocd/scripts/interface/imx-native.cfg
x openocd/share/openocd/scripts/interface/ti-icdi.cfg
x openocd/share/openocd/scripts/interface/vdebug.cfg
x openocd/share/openocd/scripts/interface/calao-usb-a9260.cfg
x openocd/share/openocd/scripts/interface/sysfsgpio-raspberrypi.cfg
x openocd/share/openocd/scripts/interface/stlink-dap.cfg
x openocd/share/openocd/scripts/interface/osbdm.cfg
x openocd/share/openocd/scripts/interface/rlink.cfg
x openocd/share/openocd/scripts/interface/beaglebone-jtag-native.cfg
x openocd/share/openocd/scripts/interface/chameleon.cfg
x openocd/share/openocd/scripts/interface/at91rm9200.cfg
x openocd/share/openocd/scripts/interface/opendous.cfg
x openocd/share/openocd/scripts/interface/raspberrypi-native.cfg
x openocd/share/openocd/scripts/interface/parport.cfg
x openocd/share/openocd/scripts/interface/openjtag.cfg
x openocd/share/openocd/scripts/interface/nds32-aice.cfg
x openocd/share/openocd/scripts/interface/stlink-v2-1.cfg
x openocd/share/openocd/scripts/interface/jlink.cfg
x openocd/share/openocd/scripts/interface/beaglebone-swd-native.cfg
x openocd/share/openocd/scripts/interface/nulink.cfg
x openocd/share/openocd/scripts/interface/rshim.cfg
x openocd/share/openocd/scripts/interface/vsllink.cfg
x openocd/share/openocd/scripts/interface/arm-jtag-ew.cfg
x openocd/share/openocd/scripts/interface/ftdi/
x openocd/share/openocd/scripts/interface/ftdi/flossjtag.cfg
x openocd/share/openocd/scripts/interface/ftdi/dp_busblaster.cfg
[... 886 lines cut ...]
 * [new branch]      release/0.8.2     -> refs/west/release/0.8.2
 * [new tag]         0.1.0             -> 0.1.0
 * [new tag]         0.2.0             -> 0.2.0
 * [new tag]         0.3.0             -> 0.3.0
 * [new tag]         0.3.1             -> 0.3.1
 * [new tag]         0.4.0             -> 0.4.0
 * [new tag]         0.5.0             -> 0.5.0
 * [new tag]         0.5.1             -> 0.5.1
 * [new tag]         0.6.0             -> 0.6.0
 * [new tag]         0.8.0             -> 0.8.0
 * [new tag]         0.8.1             -> 0.8.1
 * [new tag]         0.9.0             -> 0.9.0
 * [new tag]         0.9.1             -> 0.9.1
 * [new tag]         deprecated-master -> deprecated-master
 * [new tag]         inital-pr         -> inital-pr
 * [new tag]         old-structure     -> old-structure
HEAD is now at 9b07780 Update version to 0.9.1
HEAD is now at 9b07780 Update version to 0.9.1
ERROR: update failed for projects: hostap, lvgl

 *  The terminal process "bash '-c', '. /Users/clark/.zinstaller/env.sh && west update'" terminated with exit code: 1. 
```
I'm not sure if I am encountering a bug or if I am making a mistake somewhere in the install process. 

Please let me know what I can do to move forward. Later today I will probably retry installing on linux.

SamDCR, 2025-06-13:
Hi @csann, 

First, apologies for the delayed response, and thank you for the detailed report. 

At the moment, the extension is not fully compatible with Intel-based macOS, which is why you are encountering issues with the Debug Tools and West Workspace installations. We are working to improve support for this platform, but for now, some tools may not be correctly detected or may fail during setup on Intel macs.

If you've had a chance to try the installation on Linux, we hope it worked better. Let us know if you need any help moving forward.

Best regards,

pksublime, 2025-07-17:
FYI https://zephyr-workbench.com/docs/documentation/getting-started/getting-started-macosx lists x86