Ac6Embedded/vscode-zephyr-workbench #21: Using .conf files with --sysbuild MCUboot compile give error.

rokobertcm, 2025-06-09:
When compiling with --sysbuild and if CONF_FILE or EXTRA_CONF_FILE is specified, compiler error is given.
Reason is that conf files are also included in MCUBoot compile command.

SamDCR, 2025-06-13:
Hi @rokobertcm,

Thank you for testing our extension. You're correct, when compiling with --sysbuild and specifying a CONF_FILE or EXTRA_CONF_FILE, the configuration is applied to both MCUBoot and the custom program. We are currently working on a fix, and it will be included in an upcoming version of the extension as soon as possible.

Sorry for the issue, and thanks for bringing it to our attention.

If you have any questions or run into anything else, feel free to reach out.

Best regards,

SamDCR, 2025-06-16:
Hi @rokobertcm,

One possible workaround (until we release a fix) is to rely on sysbuild's built-in support for image-specific configuration using the sysbuild/ folder. According to the [Zephyr documentation for sysbuild](https://docs.zephyrproject.org/latest/build/sysbuild/index.html#sysbuild-file-suffix-support), if you place configuration files in:

sysbuild/
├── mcuboot.conf
└── your_app.conf

then each image will only receive the .conf matching its name. Of course, we're still working on making this smoother with the extension, but in the meantime, this approach should help you keep moving forward with your projects.

Best regards,

rokobertcm, 2025-06-17:
Thank you for quick fix solution. I'm using more than one .conf files. For example one for BLE one for ethernet one for debugging ... . It is not a must need! I can wait for fix.

Best regards,
Robert

rokobertcm, 2025-08-12:
Something new on this issue?

Best regards,
Robert