Ac6Embedded/vscode-zephyr-workbench #29: Cannot create new IAR toolchain project

thenguyenyf, 2025-07-03:
**Issue:** Create new project by the new project wizard cannot complete for IAR toolchain

**Env:**
- Windows 10
- Zephyr Workbench 1.3.23
- Zephyr SDK 0.17.2
- Zephyr IAR [9.70.1-alpha.5138](https://github.com/iarsystems/zephyr-iar/releases/tag/iar-arm-build-tools-nightly-linux-9.70.1-alpha.5138)

**Re-produce step:**
1. Import a project by new project wizard
2. Select IAR as project toolchain
3. Select the boards from board list select popup
4. Project folder will be added to workspace folder without create the .vscode configuration folder and no error message output

thenguyenyf, 2025-07-03:
> Hello [@thenguyenyf](https://github.com/thenguyenyf),
> 
> Thanks for using the extension!
> 
> I attempted to reproduce the issue using the exact environment you've provided but was unable to replicate the problem. Could you confirm if this behavior occurs with all projects or only this particular one?
> 
> Also, can you verify if the steps I'm following (shown in the attached image) match your process exactly?
> 
> Thanks!
> 
> [image]

I meet the issue while trying to import the existing project (Ex: samples/hello_world). Could you please retry this case?

thenguyenyf, 2025-07-03:
> Hello [@thenguyenyf](https://github.com/thenguyenyf),
>
> Thanks for using the extension!
>
> I attempted to reproduce the issue using the exact environment you've provided but was unable to replicate the [...]

Hello @SamDCR . I tried as your guide and its success. It seems the issue will be only occurred if using 'Import Existing Application' button. Could you please check it?

[image]

SamDCR, 2025-07-03:
Hi @thenguyenyf,

I just tried using the "Import Existing Application" option and you're right, the .vscode folder is not created. Thanks for reporting this issue.

We are working on a fix that will be available in the next version of the extension.

Glad to hear the other option worked for you, and we really appreciate your feedback!

closed 2025-07-04 by thenguyenyf