Ac6Embedded/vscode-zephyr-workbench #37: Extension almost working in Cursor

Simon33rd, 2025-08-25:
Hi Team,

Nice to see the work going on here.  The Workbench is a great tool.
I've been able to get it working on Cursor by manually loading the .vsix extension files required. 
The problem I have is with requirement to use MS C/C++ IntelliSense extension as it clashes with clangd extension which is preferred on Cursor, and by mean user of VS Code I suspect.
Is there any plans to do a Cursor friendly version of the extension?

Thanks
Simon

Simon33rd, 2025-08-26:
I managed to find a solution by installing an old version of MS C/C++ IntelliSense extension (ms-vscode.cpptools 1.23.6). It seems versions after this are not Cursor friendly.
Thanks for the heads up on Augmentcode, I hadn't come across this one.

RoyAc6, 2025-10-17:
the latest version of IntelliSense seems to work again ! we are thinking to migrate to clangd, but i am not sure if it is going to be any better ?

Simon33rd, 2025-10-18:
> do you have any experience with clangd or a setup that we can test with zephyr ?

Once the clangd extension is installed make sure it can find your clangd.exe (you might have to install it separately). look for your clangd: Path setting, for me it's:
`C:\Clangd\clangd_20.1.8\bin\clangd.exe`
The IntelliSense config in C_Cpp extension settings allows you to select clang-arm mode, and turn off IntelliSense.
My settings are:
`C_Cpp › Default: Intelli Sense Mode = clang-arm`
`C_Cpp: Intelli Sense Engine = disabled`
My settings.json file also includes:
```
 "C_Cpp.default.compilerPath": "d:\\Zephyr\\zephyr-sdk-0.17.4\\arm-zephyr-eabi\\bin\\arm-zephyr-eabi-gcc",
 "C_Cpp.default.compileCommands": "${workspaceFolder}\\build\\primary\\compile_commands.json",
  
  // Enable formatting with clang-format through clangd
  "C_Cpp.formatting": "clangFormat",
  "C_Cpp.clang_format_style": "file",
  
  // Set clangd as the default formatter
  "[cpp]": {
    "editor.defaultFormatter": "llvm-vs-code-extensions.vscode-clangd",
    "editor.formatOnSave": true
  },
  "[c]": {
    "editor.defaultFormatter": "llvm-vs-code-extensions.vscode-clangd", 
    "editor.formatOnSave": true
  }
```
My .clangd file (in project root) contains
```
# .clangd (repo root)
CompileFlags:
  # Folder that contains compile_commands.json
  CompilationDatabase: build/primary
  # Add CORE_CM4 define and include directories for clangd
  Add:
    - -DCORE_CM4
    - -Isrc
    - -Ikeys
    - -Isrc/json
    - -I../Common/Inc
    - -I../Common/Security
  # Remove two GCC-only switches that clang/clangd can't parse
  Remove:
    - -mfp16-format=ieee
    - -fno-reorder-functions
    - -fno-printf-return-value
```
I also have a .clang-format file with:
```
# Use the LLVM base style
BasedOnStyle: LLVM
# Put brackets on their own lines for control statements
BreakBeforeBraces: Allman
# Align assignments
AlignConsecutiveAssignments: true
# Always break the line after the opening parenthesis of a function call if parameters are on multiple lines
BinPackParameters: false
# Specify a tab width of 4 spaces
IndentWidth: 4
# Don't format the code between these comment markers
DisableFormat:
  - clang-format off
  - clang-format on
```
Note that you have to successfully build your project for the `compile_commands.json` file to be generated, which is need for clangd to run properly. Run command` clangd: Restart language server` for the new info to be used.